humanoid-path-planner/hpp-pinocchio #1: Grippers

jmirabel, 2016-07-18:


jmirabel, 2016-07-18:
The grippers are defined by a fixed frame in a joint. They are defined in URDF by a specific tag.
I was thinking to keep isolated (as they are currently). They could also be better integrated by using the fixed frame feature of Pinocchio. Any thoughts on this ?

nmansard, 2016-07-19:
Could you elaborate a bit? What do you mean by "as currently"?

jmirabel, 2016-07-19:
To define grippers in SRDF files, the user must specify:
- a joint (by a name),
- a placement in this joint,
- a clearance value (scalar).

The hpp-model implementation keeps these 3 information internally.

With pinocchio, there are two solution:
- keep the 3 information (non-integrated approach),
- keep a frame index (contains both the joint and placement) and a clearance value.

Actually, the integrated method is pretty easy to implement using the Frame feature of Pinocchio.
This method requires to add frame when parsing the SRDF. This is possible, right ?

closed 2016-07-21 by jmirabel