humanoid-path-planner/hpp-pinocchio #5: Joint vector

jmirabel, 2016-07-19:
About the unittest of class `Device`:
the joint vector cannot be compared for devices with anchor joints. In HPP, they appear in the joint vector. In Pinocchio, they don't.

Two solutions:
- do not compare the joint vector,
- compare it using a model without anchor joints (an embedded model if probably the best approach).

We will have to be careful in other packages as some algorithms are looping on the joint vector.

jmirabel, 2016-07-19:
If nobody complains, I will go for the solution of not comparing them.

nmansard, 2016-07-19:
Order does not matter I agree. This is a simple test to validate the random accessor []. If this test is not valid, we need to write a more complex one to validate [].

We have a test romeo model in Pinocchio without anchor joint. We might use it as well. However we will also need anchor joints for complete test later.

closed 2016-07-20 by jmirabel