humanoid-path-planner/hpp-pinocchio #7: Request_t COLLISON | DISTANCE

nmansard, 2016-07-19:
In hpp::model, all accesses to collision objects are parametrized by a Request_t to access either to COLLISION objects or DISTANCE objects. Since we decided not to keep this feature in hpp::pinocchio, the parameter is now useless. 

This also means that the signature of all these functions will change, which will be very boring in propagating the new API in hpp::core and ++. Should we either:
* put this parameter as optional (COLLISION by default) is hpp::model.
* or add temporarily a fake Request_t parameter in all these functions in hpp::pinocchio

Any opinions?

jmirabel, 2016-07-19:
A fake deprecated `Request_t` as the advantage of generating warnings.

nmansard, 2016-07-20:
Could you implement it?

closed 2016-07-20 by jmirabel