humanoid-path-planner/hpp-pinocchio #10: HPP-FCL with eigen or not

jcarpent, 2016-07-20:
This issues is related to #4. 
Currently Pinocchio uses the spatial objects from FCL to read results from collision or distance request. While hpp-model uses the eigen library.

Maybe it is time to converge on this topic and make hpp-fcl default with eigen lib and adapt the code of Pinocchio. 
What do you think?

jmirabel, 2016-07-20:
> adapt the code of Pinocchio.

I don't think there is anything to adapt in Pinocchio to make it works. It compiles and runs fine on my computer.

jcarpent, 2016-07-20:
Indeed, it works well but there are duplicated data and conversions. We can avoid them by making things uniform on both sides.

jmirabel, 2016-07-20:
IMHO, the real question is: do we accept to diverge from the official FCL ?

If yes, using the Eigenized version seems better for us.

jmirabel, 2016-07-23:
I have started to work with the Eigenized version. In hpp-constraints, there are many part where it is a bit heavy to keep full compatibility with both.

jmirabel, 2016-07-25:
I think using the Eigenized version is actually not diverging too much (if this goes up to the end):

https://github.com/flexible-collision-library/fcl/issues/96#issuecomment-234800529

nmansard, 2016-07-29:
@jcarpent mentionned that the hpp team (@jmirabel I guess) already implemented a Eigen version of hpp::fcl. Can you confirm this statement? If so, is the entire hpp project now built on a Eigen version of it?

jmirabel, 2016-08-01:
The statement is partially true only. What is done is:
- 3D vectors and 3x3 matrices are eigenized,
- quaternions are still an FCL class (because of the convention issue),
- `Transform3f` is still an FCL class, containing 2 Eigen objects (3x1 and 3x3 matrices) and 1 FCL object (quaternion).

nmansard, 2016-08-04:
And all hpp is based on that version? What about Pinocchio?

jmirabel, 2016-08-04:
Up to before the beginning of the integration, HPP was compatible with both. In the current state, it is compatible with the eigenized version.

jmirabel, 2016-09-01:
The code is now mostly (if not completely) independent of this decision. For placements, HPP (always) uses Pinocchio `SE3` and (never) hpp-fcl `Transform3f` and let Pinocchio do the conversion.

This question is not anymore a question for HPP but for Pinocchio.

closed 2016-09-01 by jmirabel