humanoid-path-planner/hpp-pinocchio #16: Add neutral configuration

nmansard, 2016-07-21:
This should be straight forward using stack-of-tasks/pinocchio#225.

nmansard, 2016-07-21:
Should we keep the neutral configuration per joint? Or the device method is enough?

jmirabel, 2016-07-21:
`hpp-rbprm` uses it:
```bash
$ grep -ER "\<neutralConfiguration\>" hpp-*/include hpp-*/src hpp-*/tests
hpp-model/include/hpp/model/device.hh:      Configuration_t neutralConfiguration () const;
hpp-model/include/hpp/model/joint.hh:      vector_t neutralConfiguration () const
hpp-pinocchio/include/hpp/pinocchio/device.hh:      Configuration_t neutralConfiguration () const;
hpp-pinocchio/include/hpp/pinocchio/joint.hh://NOTYET      vector_t neutralConfiguration () const;
hpp-model/src/device.cc:	    joint->neutralConfiguration ();
hpp-model/src/device.cc:    Configuration_t Device::neutralConfiguration () const
hpp-model/src/device.cc:          (*it)->neutralConfiguration ();
hpp-model/src/joint.cc:      neutralConfiguration_ = joint.neutralConfiguration ();
hpp-pinocchio/src/joint.cc://NOTYET    vector_t  Joint::neutralConfiguration () const {}
hpp-rbprm/src/sampling/sample.cc:            + effector->neutralConfiguration().rows();
hpp-rbprm/src/rbprm-limb.cc:        effector->robot()->currentConfiguration (effector->robot()->neutralConfiguration());
hpp-rbprm/src/ik-solver.cc:    std::size_t end = limb.effector_->rankInConfiguration()+ limb.effector_->neutralConfiguration().rows();
hpp-core/tests/relative-motion.cc:  Configuration_t q = dev->neutralConfiguration();
hpp-pinocchio/tests/tjoint.cc://NOTCHECKED      vector_t neutralConfiguration () const;
hpp-pinocchio/tests/tdevice.cc://NOTCHECKED      Configuration_t neutralConfiguration () const;
```

closed 2016-07-22 by jmirabel