humanoid-path-planner/hpp-pinocchio #17: Jacobian of COM of a subtree

jmirabel, 2016-07-22:
I was able to get the COM of a subtree but I am not able to get it's Jacobian. Did I miss something or do I have to implement it myself ?

Most of the time, the subtree I'm interested in is made of one or two subtrees so it would not be so costly to compute them separately.

jmirabel, 2016-07-22:
I had a look at the code of `computeJacobianCenterOfMass` and it seems pretty easy to compute the jacobian of a subtree from the existing code. I see two options:
- add an argument in Pinocchio (the root index of the subtree, 0 by default),
- do my own algorithm in hpp-pinocchio.

Are the Pinocchio developpers ( @nmansard @jcarpent ) interested in the option 1 ?

jcarpent, 2016-07-22:
I do not think that it is trivial.
Just because the jacobian of the com subtree starting at j_id is influenced both by its parents and its children.

I'm adding the subtree information in model to support it.
Do you agree?

jmirabel, 2016-07-22:
Great, thanks !

About the math: I'm probably wrong but isn't it the weighed sum of the derivative of joint placements ? How can it be influenced by its parents ?

jcarpent, 2016-07-22:
The velocity of the com of a certain joint is influences by the velocity of its supporting joints, like the jacobians.

But, in the case of the com subtree, it also influenced by the CoM of its children, by definition. I'm not sure if I'm clear. 
You may have look to the algo centerOfMass(model,data,q,v) which computes both the CoMs and their velocities with a forward and a backward steps.

jmirabel, 2016-07-22:
Got it, thanks !

jmirabel, 2016-07-23:
I came up with that: https://github.com/humanoid-path-planner/hpp-pinocchio/blob/master/src/center-of-mass-computation/algorithm.cc

Except for some problems related to https://github.com/stack-of-tasks/pinocchio/pull/250, it was actually pretty easy. The comparison with HPP works fine.

jcarpent, 2016-07-24:
Again, the velocity of the CoM does depends on both the DoFs of the subtree and also the DoFs supporting the subtree.

jcarpent, 2016-07-24:
This means that vcom[subtree] =Jcom_subtree * q_dot

jmirabel, 2016-07-24:
Yes, I realized that today. My unit test was working because I was using
the root joint as root of the subtree. Not very smart...

I think I know how to modify it without too much effort to get the right
behavior. We can discuss it tomorrow.

jcarpent, 2016-07-24:
I implemented the correct version if you want
https://github.com/jcarpent/pinocchio/tree/topic/subtree-com
I think it is interested to move it inside Pinocchio directly. I can also add a method which uses the result of jacobianCenterOfMass without further computations. Are you interested?

nmansard, 2016-07-26:
Related to #28 
This PR should close the issue if accepted.

closed 2016-07-26 by nmansard