humanoid-path-planner/hpp-pinocchio #19: Calls to FCL

jmirabel, 2016-07-23:
With the API of hpp-model, a distance request to FCL was:
```cpp
fcl::distance(obj1->fcl(), obj2->fcl(), request, result)
```
With hpp-pinocchio, we have:
```cpp
fcl::distance(obj1->fcl().collisionGeometry().get(), obj1->getFclTransform(), 
              obj2->fcl().collisionGeometry().get(), obj1->getFclTransform(),
              request, result)
```
which is not very convenient.

This is because pinocchio do not use the transformation inside the `fcl::CollisionObject` (I think but I'm not sure). The API of `CollisionObject` to get a more handy and readable code.

nmansard, 2016-07-25:
Related to stack-of-tasks/pinocchio#253

nmansard, 2016-07-29:
Solved by stack-of-tasks/pinocchio#280

closed 2016-07-29 by nmansard