humanoid-path-planner/hpp-pinocchio #23: Porting notes

jmirabel, 2016-07-25:
Here is a list of points that should be taken care of:

- Typedefs:
  - `Transform3f` is now a `se3::SE3` object. Unlike `fcl::Transform3f`, the constructor does not initialize itself to identity. `transform` becomes `act`.
  - `vector3_t` becomes an Eigen vector
  - `JointPtr_t` is a boost shared_ptr

- Comparing joint position:
```cpp
m2p::se3(
jointModel->currentTransformation() * jointModel->linkInJointFrame()
).isApprox(
jointPinocchio->currentTransformation() * jointPinocchio->robot()->model().getFramePlacement(jointModel->linkName())
)
```

jcarpent, 2016-07-25:
For the second point, I think we need to discuss more. I think we might change the convention of local and global in Pinocchio. Currently, in Pinocchio:
- local means it is the velocity of the centre of joint expressed in the joint frame
- global means it is the velocity of the point coinciding with the centre of the world, attached to the joint frame but expressed in the world frame.

I suggest to change global using the same convention than HPP, i.e. it is the velocity of the centre of the joint expressed in the global frame. @jmirabel @nmansard What do you think?

Finally, the best solution ever is to add the notion of MotionSet inside Pinocchio. The idea is to have a class which mimics the current Matrix6x on which SE3 and others can act on. But this for future ... ;)

jmirabel, 2016-07-25:
In an absolute view, I do not see what are the pros and cons of any of the options (neither the point, nor the frame). I guess there are some computational advantages, but I'm not sure.

nmansard, 2016-07-25:
I do not think that changing Jacobians in Pinocchio is a good idea. We are following the basic convention with the motion/force elements that the point of expression of the vector field is always the center of the frame. I like this convention, even if it is not the classical way of doing the maths.

jcarpent, 2016-07-26:
I think is not changing the convention, just the return of getJacobian<global>. I think the user can expect to still work around the same point. Not having a function which returns two quantities expressed around two different points/frames.

nmansard, 2016-07-26:
The point is that the jacobian is a (linear) application transforming configuration velocities vq to cartesian "motion-6" velocities nu. We must follow this convention, otherwise it becomes a gigantic mathematical mess.
As nicely stated above, the good idea is to see the Jacobian as a collection of Motion elements. Having a motion expressed in a frame A at the center of another frame B is not part of how we can work with Motion.

closed 2018-02-21 by jmirabel