humanoid-path-planner/hpp-pinocchio #24: Current forward declaration

jmirabel, 2016-07-25:
This is a list of the current forward declared types. Any opinion on any of them ?

- [ ]  HPP_PREDEF_CLASS (Body);
- [ ]  HPP_PREDEF_CLASS (CollisionObject);
- [ ]  HPP_PREDEF_CLASS (Device);
- [ ]  //DEPREC HPP_PREDEF_CLASS (DistanceResult);
- [ ]  HPP_PREDEF_CLASS (HumanoidRobot);
- [ ]  HPP_PREDEF_CLASS (Joint);
- [ ]  HPP_PREDEF_CLASS (JointConfiguration);
- [ ]  HPP_PREDEF_CLASS (Gripper);
- [ ]  enum Request_t {COLLISION, DISTANCE};
- [ ]  typedef double value_type;
- [ ]  typedef Eigen::Matrix <value_type, Eigen::Dynamic, 1> vector_t;
- [ ]  typedef vector_t Configuration_t;
- [ ]  typedef Eigen::Ref <const Configuration_t> ConfigurationIn_t;
- [ ]  typedef Eigen::Ref <Configuration_t> ConfigurationOut_t;
- [ ]  typedef boost::shared_ptr <Configuration_t> ConfigurationPtr_t;
- [ ]  typedef Eigen::Ref <const vector_t> vectorIn_t;
- [ ]  typedef Eigen::Ref <vector_t> vectorOut_t;
- [ ]  typedef Eigen::Matrix<value_type, Eigen::Dynamic, Eigen::Dynamic> matrix_t;
- [ ]  typedef Eigen::Ref <matrix_t> matrixOut_t;
- [ ]  typedef matrix_t::Index size_type;
- [ ]  typedef fcl::Matrix3f matrix3_t;
- [ ]  typedef fcl::Vec3f vector3_t;
- [ ]  typedef se3::Data::Matrix6x JointJacobian_t;
- [ ]  typedef se3::Data::Matrix3x ComJacobian_t;
- [ ]  typedef Eigen::Block <JointJacobian_t, 3, Eigen::Dynamic> HalfJointJacobian_t;
- [ ]  struct JointVector;
- [ ]  typedef JointVector JointVector_t;
- [ ]  typedef boost::shared_ptr<Body> BodyPtr_t;
- [ ]  typedef boost::shared_ptr<const Body> BodyConstPtr_t;
- [ ]  //typedef std::vector<BodyPtr_t> BodyVector_t;
- [ ]  typedef fcl::CollisionObject * fclCollisionObjectPtr_t;
- [ ]  typedef const fcl::CollisionObject * fclConstCollisionObjectPtr_t;
- [ ]  //DEPREC typedef std::list <CollisionObjectPtr_t> ObjectVector_t;
- [ ]  typedef boost::shared_ptr <Device> DevicePtr_t;
- [ ]  typedef boost::shared_ptr <const Device> DeviceConstPtr_t;
- [ ]  typedef std::vector <se3::DistanceResult> DistanceResults_t;
- [ ]  typedef boost::shared_ptr <HumanoidRobot> HumanoidRobotPtr_t;
- [ ]  typedef boost::shared_ptr<Joint> JointPtr_t;
- [ ]  typedef boost::shared_ptr<const Joint> JointConstPtr_t;
- [ ]  typedef boost::shared_ptr <Gripper> GripperPtr_t;
- [ ]  typedef std::vector <GripperPtr_t> Grippers_t;
- [ ]  //DEPREC typedef fcl::Transform3f Transform3f;
- [ ]  typedef se3::SE3 Transform3f;
- [ ]  typedef boost::shared_ptr<se3::Model>               ModelPtr_t;
- [ ]  typedef boost::shared_ptr<const se3::Model>         ModelConstPtr_t;
- [ ]  typedef boost::shared_ptr<se3::Data>                DataPtr_t;
- [ ]  typedef boost::shared_ptr<const se3::Data>          DataConstPtr_t;
- [ ]  typedef boost::shared_ptr<se3::GeometryModel>       GeomModelPtr_t;
- [ ]  typedef boost::shared_ptr<const se3::GeometryModel> GeomModelConstPtr_t;
- [ ]  typedef boost::shared_ptr<se3::GeometryData>        GeomDataPtr_t;
- [ ]  typedef boost::shared_ptr<const se3::GeometryData>  GeomDataConstPtr_t;

jmirabel, 2016-07-25:
`JointJacobian_t` and `ComJacobian_t`: I do not really like them not being dependant on `value_type` like the rest. But I guess we have no choice yet. Correct ?

nmansard, 2016-07-25:
We can have a fake dependency, which would be valid as far as value_type is double. 
Pinocchio can not be float (yet).

nmansard, 2016-07-25:
I do not like to have explicit include in forward (on pinocchio/model, in particular). I think it would be a good idea to have the Jacobian being defined from Eigen directly, possibly with value_type.

jmirabel, 2016-07-25:
So this would be better:
```cpp
typedef Eigen::Matrix<value_type, 6, Eigen::Dynamic> Jacobian_t;
typedef Eigen::Matrix<value_type, 3, Eigen::Dynamic> ComJacobian_t;
```

nmansard, 2016-07-25:
I like it. Maybe we could add a static assert in the corresponding 
methods, to check that both types are compatible?

jmirabel, 2016-07-25:
See a5474bb4c7efba9e1ad67c5600d37a75cd6ad6e3

jmirabel, 2016-07-25:
Is there something in Eigen to check the "compatibility" for us ?

jcarpent, 2016-07-26:
As far as I, know there is no such predefined macro. You must define your own and us boost::is_same feature.

nmansard, 2016-07-26:
There are static asserts in Eigen to check the size of two matrices and whether these sizes are compatible. As proposed by @jcarpent you can also check whether the scalar types are the same. This might only be a warning.

closed 2016-08-19 by jmirabel