humanoid-path-planner/hpp-pinocchio #26: Parsing of URDF/SRDF

jmirabel, 2016-07-25:
`Device::geomData_` should contain a default list of collision pairs:
- build from the URDF/SRDF when parsing the files,
- updated by the user who wants to disable some pairs.

nmansard, 2016-07-25:
I like it. We need to put that in pinocchio::Model. Could you propose a PR in Pinocchio with that implemented?

jcarpent, 2016-07-26:
I think there is already such features in Pinocchio. You can activate all collisions pairs in a row and then use SRDF to deactivate them.

nmansard, 2016-07-26:
The idea of @jmirabel is to have a initial constant version of collision pairs in the model, that would be used to initialize the data by default. Then possibly some algorithms would modify the data instance of collision pairs, but they can always be restored to the default value without re-parsing the srdf.

jmirabel, 2016-07-27:
The current missing features to have the same API as `hpp-model-urdf` are:
- the ability to add a prefix: the workaround is to post-process the model and rename joints and links,
- the ability to append a model to another (at the world would be enough).

Both points are required for the manipulation stack when merging models together.

jcarpent, 2016-07-27:
For the first point, the prefix can be added a posteriori, after building the model.
For the second point, I think the class Model in Pinocchio has to handle this scenario, with a tailored algorithm to reorganise joints and obtained aligned subtree.

jmirabel, 2016-07-27:
The algorithm is not required for now as I have always been adding robots at the world frame (joint 0). I guess compactness is automatically preserved when doing so.

jcarpent, 2016-07-27:
Then, you cannot use some algo, like getJacobianFromCrba and so on ...
Just be aware of such limitations of concatenating some models together.

nmansard, 2016-07-27:
This discussion is too long. Let's modify the URDF parser only for now. Currently the signature is:
    inline Model buildModel (const std::string & filename,
                             const bool verbose = false) throw (std::invalid_argument);
It should become
    inline Model& buildModel (const std::string & filename,
                             Model & model, const bool verbose = false) throw (std::invalid_argument);
For compatibility reason, the initial signature can be kept with a forward on the new signature. Mark it as deprecated.

I dont have the time to implement it today. I would likely have time tomorrow morning. 
@jmirabel : could you make the change yourself? I don't think it is difficult. The change would be in conflict with a part of the current work of @jcarpent (not merged yet), but merging later wont be a big deal. 
Alternatively, would somebody else (@fvalenza?) be available for doing the change?

Following of this discussion about how to implement it in pinocchio should be posted on the corresponding issue of pinocchio.

jmirabel, 2016-07-27:
I won't have time to do it before Monday as I'm leaving today.

jcarpent, 2016-07-27:
@nmansard I prefer to change the name of your proposition. Keep `buildModel` as it is now and create a new function `appendToModel`. In this way, we keep the compatibility with previous code for any project.

nmansard, 2016-07-29:
I kept the former parser signature as deprecated and added a new function with same name that is able to append models. 
I moved collision pairs to geomModel and added a mask of active pairs in geomData.
I think that it closes the issue.

closed 2016-07-29 by nmansard