humanoid-path-planner/hpp-pinocchio #27: Handling of collision geometries

jmirabel, 2016-07-26:
`GeometryData` contains a vector of `GeometryObject` built from the robot model. There are two points that should be discussed now, with a priority order:

1. Add obstacles,
2. Add robots.

### Add obstacles
The obstacles will be added to the `GeometryData` of the device. So far, `GeometryData` has a vector of `CollisionObject`. To handle static objects, we can:
- add a `GeometryObject` whose parent joint is 0. Any time the geometry placement is updated, we apply a `Data::oMi[0] == SE3::Identity()` transformation to the geometry,
- add them at the end and keep the index of the last moving geometry. In other words, keep the vector so that the moving geometries are first. This may cause an issue later, when loading several robots and environments in an undefined order.
- add a second list of static geometries.

The corresponding `GeometryData` has to be updated accordingly but this won't be an issue as the object having a `GeometryData` are already notified when an obstacle is added.

### Append a robot
I haven't really thought of this but it would be good to start the discussion. Is there a way of:
- either appending a model to another,
- or read a URDF directly into a model, providing the joint to which the model from URDF must be appended ?

nmansard, 2016-07-26:
Add obstacles: the initial concept of Pinocchio is to add "obstacles" as object of the universe, ie whose parent is joint 0. The SE3 update loops always (should) treat 0 as a special case, generally by starting the joint loop at index 1. Do you see any problem with this solution? 

Append robot: currently, the parsing methods return a new Model. They should be modified by simply being given a Model reference as argument, that they fill. A new robot would be added to the "universe" without modifying anything else. If you want, you can open a "todo" issue about this specific topic. 
We also discussed the possibility to have the newly-parsed robot attached not to the universe but to an existing joint, by adding an additional argument to the parsing function "rootId" whose default value is 0. It would also work without further modifications. However, this would break the hypothesis of "compactness" of the kinematic tree, and would therefore imply to write a re-ordering algorithm. This is not difficult but require more work.
I suggest to implement the first point quickly, but to let the second point as an open issue as it is very unlikely that anybody requires this feature in a near future.
Please post any further comment about the issue "append robot" in a dedicated thread.

jmirabel, 2016-07-26:
I do not see any problem. I was just afraid the obstacle position would be computed every time `updateGeometryPlacement` is called.

nmansard, 2016-07-26:
Double check: yes it is (pinocchio/src/algorithm/geometry.hxx:46). We simply need to put a branching to avoid re-placement when parent = 0.

nmansard, 2016-07-29:
Add obstacles is solved by this discussion. Not updating their placement is solved by stack-of-tasks/pinocchio#280.
Append robot is solved by stack-of-tasks/pinocchio#280.
I close this issue.

closed 2016-07-29 by nmansard