humanoid-path-planner/hpp-pinocchio #32: tconfiguration: boost check error

nmansard, 2016-07-29:
There is a reported bug in Pinocchio (stack-of-tasks/pinocchio#278) in the joint difference and interpolation.
Consequently, the corresponding configuration functions of hpp:;pinocchio are also bugged. A direct consequence is that the test tconfiguration is compiling but not running anymore.

We expect the bug to be fixed in the current week. Currently, the numerical value are close to be correct, so it should be possible to implement functional tests in e.g. hpp::core.

jmirabel, 2016-08-25:
@jcarpent Could you confirm the bug is solved in Pinocchio now ?

jcarpent, 2016-08-25:
Yes, indeed. See stack-of-tasks/pinocchio/issues/278 for further details.

closed 2016-08-25 by jmirabel