humanoid-path-planner/hpp-pinocchio #34: Robot model: joint, link and geometry

jmirabel, 2016-08-03:
I have come across a dilemma which has not been stated clearly. The following terms are not completely defined:
- joint: what does it contains: one body, several bodies, geometry objects?
- body: composed of one or more geometry object ? Just a frame with no way to recover the geometry object? 
- geometry object: what is its parent?

After all the issues I came across, I think the most suited is:
- geometry object must know the body they belong to. Otherwise, because Pinocchio removes anchor joints, it is not possible to handle properly the disable_collision tag.
- bodies are frames, with one or more geometry object.
- joints have one or more body.

Is there somebody who sees a problem
- having several bodies in a joint,
- making bodies as parent of geometry objects, instead of joints?

This implies a few changes in Pinocchio.

nmansard, 2016-08-04:
In Pinocchio there is no notion of bodies. In general , bodies are elements (mass, name, geometry, property, etc) attached to joints. We only store the joints. I think it is the good way of doing it (i.e. A kinematic tree is a tree of joints decorated with various elements).
Could you discuss more the problem you are mentioning about disable collision tag?

jmirabel, 2016-08-04:
In case the URDF is something like (L == Link and J == Joint):
L0
| --> J0 --> L1 --> J1 --> L2
| --> J3 --> L4 -> J4

If J1 is a anchor joint, then L1 and L2 will be merged and it becomes impossible to disable collisions between L4 and L1 (or L4 and L2). This is because it is not possible to associate geometry objects of a joint with links of that joint, when there are several joints.

Independently of this issue, I think the notion of body is not necessary and I agree with you, the kinematic tree should only contain joints. However, I also think it is very convenient when dealing with geometries. A body represents an object of the world with possibly several geometry objects. It can be seen as a group of geometry objects.

nmansard, 2016-08-05:
Isn't it possible to specify pair of objects instead of pairs of bodies in the srdf?
I am against adding bodies in Pinocchio, and not in favor of adding a big number of group of bodies to mimc bodies.

florent-lamiraux, 2016-08-05:
I think we should keep a compatibility with existing srdf and be able to parse command
`    <disable_collisions link1="" link2="" />`
We only need to keep somewhere a mapping between link names and vectors of objects.

jmirabel, 2016-08-05:
I agree with @florent-lamiraux. First geometry objects have no name in URDF. Second we shouldn't change the SRDF specifications, otherwise each time we consider a new robot, we have to rework the SRDF.

jmirabel, 2016-08-09:
This question is currently blocking further developments so we __have to__ decide.

@jcarpent @nmansard @florent-lamiraux :
Currently, the parent of `GeometryObject`s is a joint ID. The solution I propose is to use a frame ID instead of a joint ID.

This implies that the GeometryModel needs a reference to the model in some places.

nmansard, 2016-08-30:
The solution proposed by @jmirabel is to add a 'parentFrame' in geometry objects of pinocchio, to remember the body they are attached to. I think this solution is closing the topic. Is it not?

jmirabel, 2016-08-30:
It is. I'm only waiting for these changes to me merged.

closed 2018-03-25 by jmirabel