humanoid-path-planner/hpp-pinocchio #40: Change JointPtr_t to avoid malloc

jmirabel, 2016-08-26:
Currently, `JointPtr_t` is `boost::shared_ptr<Joint>`. I think it could simply be something like:
```cpp
class JointPtr_t
{
DevicePtr_t d;
JointIndex jid;

Joint& operator-> ();
Joint& operator* ();
};
```

nmansard, 2016-08-30:
The smart pointers design is inherited from hpp-model for all objects except for joints. We inserted smart pointers for joints to enforce symmetry in the design. I still think that this symmetry objective is a good choice. Why did you change your mind?

jmirabel, 2016-08-30:
`JointPtr_t` used to be a `Joint*` and not a shared pointer but that's not my reason.

1. When updating hpp-core to use hpp-pinocchio, I had a loop which was creating many `JointPtr_t` from `new Joint` implicitly. The result was that the whole code was __really__ slow. It didn't take me long to know where the issue was because I know what functions allocate new `Joint`. However, this behaviour might be surprising for people not having such knowledge.

2. IMHO, there should not be a `malloc` when no actual object is created.

closed 2026-08-25 by nim65s