humanoid-path-planner/hpp-pinocchio #72: Tests are not passing on master

nim65s, 2018-10-04:
Hi,

While trying to make a new release from master, I got this issue:
```
╰─>$ ./tests/tconfiguration 
Running 6 test cases...
tconfiguration: /usr/include/eigen3/Eigen/src/Core/DenseCoeffsBase.h:396: Eigen::DenseCoeffsBase<Derived, 1>::Scalar& Eigen::DenseCoeffsBase<Derived, 1>::operator[](Eigen::Index) [with Derived = Eigen::Matrix<double, -1, 1>; Eigen::DenseCoeffsBase<Derived, 1>::Scalar = double; Eigen::Index = long int]: Assertion `index >= 0 && index < size()' failed.
unknown location(0): fatal error in "difference_and_distance": signal: SIGABRT (application abort requested)
/local/users/gsaurel/humanoid-path-planner/hpp-pinocchio/tests/tconfiguration.cc(155): last checkpoint
tconfiguration: /usr/include/eigen3/Eigen/src/Core/DenseCoeffsBase.h:396: Eigen::DenseCoeffsBase<Derived, 1>::Scalar& Eigen::DenseCoeffsBase<Derived, 1>::operator[](Eigen::Index) [with Derived = Eigen::Matrix<double, -1, 1>; Eigen::DenseCoeffsBase<Derived, 1>::Scalar = double; Eigen::Index = long int]: Assertion `index >= 0 && index < size()' failed.
unknown location(0): fatal error in "interpolate_and_integrate": signal: SIGABRT (application abort requested)
/local/users/gsaurel/humanoid-path-planner/hpp-pinocchio/tests/tconfiguration.cc(233): last checkpoint

*** 2 failures detected in test suite "tdevice"
```

Am I missing something ?

nim65s, 2018-10-04:
It seems that this test is not launched by hpp-doc:

```
┌─[gsaurel@akasan:$DEVEL_HPP_DIR/src/hpp-pinocchio/build-rel]-[16:34:15]─[G:master|✔]
└─>$ make test
[…]
100% tests passed, 0 tests failed out of 3
┬─[gsaurel@akasan:~/local/humanoid-path-planner/hpp-pinocchio/build]─[16:38:48]─[G:master|✔]
╰─>$ make test
[…]
83% tests passed, 1 tests failed out of 6
```

Anyway, I can still make the release from the build in `$DEVEL_HPP_DIR`

jmirabel, 2018-10-04:
I would be curious to get a backtrace. I cannot reproduce this on my computer.