humanoid-path-planner/hpp-pinocchio #88: fatal error in computeAABB on 18.04 / debug

nim65s, 2019-03-19:
https://gepgitlab.laas.fr/gsaurel/hpp-pinocchio/-/jobs/15303 raises the following error in the `device` test:
```
unknown location(0): fatal error: in "computeAABB": std::invalid_argument: Expected vector of size 3 or 7, got size 140199579517522
/root/robotpkg/math/hpp-pinocchio/work.runner-991216bb-project-186-concurrent-3/hpp-pinocchio-4.4.0/tests/device.cc(44): last checkpoint

*** 1 failure is detected in the test module "tframe"
```

This test correctly passes on 18.04 / release and on 16.04.

jmirabel, 2019-03-19:
I think @florent-lamiraux and I fixed it. It must be on @florent-lamiraux's fork

jmirabel, 2019-06-07:
Fixed by fd258b23f10e273f29779f19caca2f516181d2b9

closed 2019-06-07 by jmirabel