humanoid-path-planner/hpp-pinocchio #90: Compatibility issue between hpp::pinocchio::Model and pinocchio::Model

olivier-roussel, 2019-04-04:
While updating hpp-pinocchio from v4.2.0 to v4.4.0 and pinocchio v1.3.1 to v2.1.0, the following minimalist example does not compile anymore:
```
hpp::pinocchio::DevicePtr_t robot = hpp::pinocchio::Device::create("manipulator"); 
hpp::pinocchio::urdf::loadModel(...); // load some URDF file

pinocchio::Data data(robot->model()); // Broken

Eigen::VectorXd q = Eigen::VectorXd::Zero(robot->model().nq);
pinocchio::framesForwardKinematics(robot->model(), data, q);  // Broken
```
From the output, it seems that could not cast hpp::pinocchio::Model to pinocchio::Model anymore, but may be I am missing something:
```

[...]/profiles/hpp_ros_kinetic-x64-ubuntu16/src/sandbox_arm_plan/src/main.cpp: In function ‘int main(int, char**)’:
[...]/profiles/hpp_ros_kinetic-x64-ubuntu16/src/sandbox_arm_plan/src/main.cpp:51:38: error: no matching function for call to ‘pinocchio::DataTpl<double>::DataTpl(hpp::pinocchio::Model&)’
   pinocchio::Data data(robot->model());
                                      ^
In file included from [...]/profiles/hpp_ros_kinetic-x64-ubuntu16/install/include/pinocchio/multibody/data.hpp:341:0,
                 from [...]/profiles/hpp_ros_kinetic-x64-ubuntu16/src/sandbox_arm_plan/src/main.cpp:43:
[...]/profiles/hpp_ros_kinetic-x64-ubuntu16/install/include/pinocchio/multibody/data.hxx:23:10: note: candidate: pinocchio::DataTpl<Scalar, Options, JointCollectionTpl>::DataTpl(const Model&) [with _Scalar = double; int _Options = 0; JointCollectionTpl = pinocchio::JointCollectionDefaultTpl; pinocchio::DataTpl<Scalar, Options, JointCollectionTpl>::Model = pinocchio::ModelTpl<double>]
   inline DataTpl<Scalar,Options,JointCollectionTpl>::
          ^
[...]/profiles/hpp_ros_kinetic-x64-ubuntu16/install/include/pinocchio/multibody/data.hxx:23:10: note:   no known conversion for argument 1 from ‘hpp::pinocchio::Model {aka pinocchio::ModelTpl<double, 0, hpp::pinocchio::JointCollectionTpl>}’ to ‘const Model& {aka const pinocchio::ModelTpl<double>&}’
In file included from [...]/profiles/hpp_ros_kinetic-x64-ubuntu16/src/sandbox_arm_plan/src/main.cpp:43:0:
[...]/profiles/hpp_ros_kinetic-x64-ubuntu16/install/include/pinocchio/multibody/data.hpp:25:10: note: candidate: pinocchio::DataTpl<double>::DataTpl(const pinocchio::DataTpl<double>&)
   struct DataTpl
          ^
[...]/profiles/hpp_ros_kinetic-x64-ubuntu16/install/include/pinocchio/multibody/data.hpp:25:10: note:   no known conversion for argument 1 from ‘hpp::pinocchio::Model {aka pinocchio::ModelTpl<double, 0, hpp::pinocchio::JointCollectionTpl>}’ to ‘const pinocchio::DataTpl<double>&’
[...]/profiles/hpp_ros_kinetic-x64-ubuntu16/install/include/pinocchio/multibody/data.hpp:25:10: note: candidate: pinocchio::DataTpl<double>::DataTpl(pinocchio::DataTpl<double>&&)
[...]/profiles/hpp_ros_kinetic-x64-ubuntu16/install/include/pinocchio/multibody/data.hpp:25:10: note:   no known conversion for argument 1 from ‘hpp::pinocchio::Model {aka pinocchio::ModelTpl<double, 0, hpp::pinocchio::JointCollectionTpl>}’ to ‘pinocchio::DataTpl<double>&&’
[...]/profiles/hpp_ros_kinetic-x64-ubuntu16/src/sandbox_arm_plan/src/main.cpp:54:61: error: no matching function for call to ‘framesForwardKinematics(hpp::pinocchio::Model&, pinocchio::Data&, Eigen::VectorXd&)’
   pinocchio::framesForwardKinematics(robot->model(), data, q); 
```
I am using pinocchio v2.1.0 (commit db8fe0ad ) and hpp-pinocchio v4.4.0 (commit d9cc0294 ), same commits than the one used in last hpp_benchmark ( https://github.com/humanoid-path-planner/hpp_benchmark/blob/master/2019-03-13/commit )

jmirabel, 2019-04-04:
They aren't compatible because the JointCollectionTpl is different. You must use the hpp::pinocchio::Data typedef.

olivier-roussel, 2019-04-04:
You mean replacing: 
```
pinocchio::Data data(robot->model());
```
by
```
hpp::pinocchio::Data data(robot->model());
```
?

jmirabel, 2019-04-04:
Yes.

jmirabel, 2019-04-04:
Same type as `the device->data()̀` returns.

olivier-roussel, 2019-04-04:
Already tried. Got this output:
```
n file included from [...]/profiles/hpp_ros_kinetic-x64-ubuntu16/install/include/pinocchio/multibody/model.hpp:17:0,
                 from [...]/profiles/hpp_ros_kinetic-x64-ubuntu16/src/sandbox_arm_plan/src/glviewer_objects/graphical_robot.hpp:16,
                 from [...]/profiles/hpp_ros_kinetic-x64-ubuntu16/src/sandbox_arm_plan/src/controller.hpp:8,
                 from [...]/profiles/hpp_ros_kinetic-x64-ubuntu16/src/sandbox_arm_plan/src/viewer.hpp:19,
                 from [...]/profiles/hpp_ros_kinetic-x64-ubuntu16/src/sandbox_arm_plan/src/main.cpp:1:
[...]/profiles/hpp_ros_kinetic-x64-ubuntu16/install/include/pinocchio/multibody/joint/joint-generic.hpp: In instantiation of ‘struct pinocchio::JointDataTpl<double, 0, hpp::pinocchio::JointCollectionTpl>’:
[...]/profiles/hpp_ros_kinetic-x64-ubuntu16/install/include/pinocchio/multibody/data.hxx:92:42:   required from ‘pinocchio::DataTpl<Scalar, Options, JointCollectionTpl>::DataTpl(const Model&) [with _Scalar = double; int _Options = 0; JointCollectionTpl = hpp::pinocchio::JointCollectionTpl; pinocchio::DataTpl<Scalar, Options, JointCollectionTpl>::Model = pinocchio::ModelTpl<double, 0, hpp::pinocchio::JointCollectionTpl>]’
[...]/profiles/hpp_ros_kinetic-x64-ubuntu16/src/sandbox_arm_plan/src/main.cpp:51:43:   required from here
[...]/profiles/hpp_ros_kinetic-x64-ubuntu16/install/include/pinocchio/multibody/joint/joint-generic.hpp:69:10: error: invalid use of incomplete type ‘struct hpp::pinocchio::JointCollectionTpl<double, 0>’
   struct JointDataTpl
          ^
In file included from [...]/profiles/hpp_ros_kinetic-x64-ubuntu16/src/sandbox_arm_plan/src/glviewer_objects/graphical_robot.hpp:15:0,
                 from [...]/profiles/hpp_ros_kinetic-x64-ubuntu16/src/sandbox_arm_plan/src/controller.hpp:8,
                 from [...]/profiles/hpp_ros_kinetic-x64-ubuntu16/src/sandbox_arm_plan/src/viewer.hpp:19,
                 from [...]/profiles/hpp_ros_kinetic-x64-ubuntu16/src/sandbox_arm_plan/src/main.cpp:1:
[...]/profiles/hpp_ros_kinetic-x64-ubuntu16/install/include/hpp/pinocchio/fwd.hh:55:54: note: declaration of ‘struct hpp::pinocchio::JointCollectionTpl<double, 0>’
     template <typename _Scalar, int _Options> struct JointCollectionTpl;
                                                      ^
In file included from [...]/profiles/hpp_ros_kinetic-x64-ubuntu16/install/include/pinocchio/multibody/model.hpp:17:0,
                 from [...]/profiles/hpp_ros_kinetic-x64-ubuntu16/src/sandbox_arm_plan/src/glviewer_objects/graphical_robot.hpp:16,
                 from [...]/profiles/hpp_ros_kinetic-x64-ubuntu16/src/sandbox_arm_plan/src/controller.hpp:8,
                 from [...]/profiles/hpp_ros_kinetic-x64-ubuntu16/src/sandbox_arm_plan/src/viewer.hpp:19,
                 from [...]/profiles/hpp_ros_kinetic-x64-ubuntu16/src/sandbox_arm_plan/src/main.cpp:1:
[...]/profiles/hpp_ros_kinetic-x64-ubuntu16/install/include/pinocchio/multibody/joint/joint-generic.hpp:81:56: error: invalid use of incomplete type ‘pinocchio::JointDataTpl<double, 0, hpp::pinocchio::JointCollectionTpl>::JointCollection {aka struct hpp::pinocchio::JointCollectionTpl<double, 0>}’
     typedef typename JointCollection::JointDataVariant JointDataVariant;
                                                        ^
In file included from [...]/profiles/hpp_ros_kinetic-x64-ubuntu16/src/sandbox_arm_plan/src/glviewer_objects/graphical_robot.hpp:15:0,
                 from [...]/profiles/hpp_ros_kinetic-x64-ubuntu16/src/sandbox_arm_plan/src/controller.hpp:8,
                 from [...]/profiles/hpp_ros_kinetic-x64-ubuntu16/src/sandbox_arm_plan/src/viewer.hpp:19,
                 from [...]/profiles/hpp_ros_kinetic-x64-ubuntu16/src/sandbox_arm_plan/src/main.cpp:1:
[...]/profiles/hpp_ros_kinetic-x64-ubuntu16/install/include/hpp/pinocchio/fwd.hh:55:54: note: declaration of ‘pinocchio::JointDataTpl<double, 0, hpp::pinocchio::JointCollectionTpl>::JointCollection {aka struct hpp::pinocchio::JointCollectionTpl<double, 0>}’
     template <typename _Scalar, int _Options> struct JointCollectionTpl;
```

jmirabel, 2019-04-04:
Add this:
`#include <hpp/pinocchio/joint-collection.hh>`

jmirabel, 2019-04-04:
I admit that it isn't easy to find the fix given the error message...

olivier-roussel, 2019-04-04:
Yes, templates...

olivier-roussel, 2019-04-04:
The missing header `#include <hpp/pinocchio/joint-collection.hh>`was guilty. Thanks. Wouldn't it be better to hide this to the user ? Looks like more an implementation detail rather than an API change.

jmirabel, 2019-04-04:
I think I didn't do it in order to reduce header dependencies and make re-compilation time shorter.

This header includes many stuff from Pinocchio. Pinocchio being a header only library, a single change in it would trigger a recompilation of almost everything. But that's only for developers convenience.

olivier-roussel, 2019-04-04:
Ok, I guess this make sense with respect to the expected audience of the software. Thanks again for the quick fix.

closed 2019-04-04 by olivier-roussel