humanoid-path-planner/hpp-pinocchio #95: Build fail with Pinocchio v2.1.5 / Eigenpy v1.5.6

nim65s, 2019-07-17:
Hi,

I tried to update robotpkg with eigenpy 1.5.6 & pinocchio 2.1.5, and building hpp-pinocchio v5.6.1 failed, with the following errors:

```
…/hpp-pinocchio-4.6.1/src/device.cc:380:67: error: no type named 'type' in 'struct hpp::pinocchio::RnxSOnLieGroupMap::operation<pinocchio::JointModelMimic<pinocchio::JointModelRevoluteTpl<double, 0, 2> > >'
typedef typename LieGroupTpl::operation<JointModel>::type LG_t;
^
…/hpp-pinocchio-4.6.1/src/device.cc:393:44: error: invalid use of member function (did you forget the '()' ?)
vector3_t min = data.M.translation();
^
…/hpp-pinocchio-4.6.1/src/device.cc:398:44: error: invalid use of member function (did you forget the '()' ?)
vector3_t max = data.M.translation();
^
```

jmirabel, 2019-07-17:
Thanks for the report. I'll have a look.

jmirabel, 2019-07-17:
01c147f should solve your issue.
If you encounter an issue regarding JointModelComposite in other packages, you should replace class `::pinocchio::JointModelComposite` by `hpp::pinocchio::JointModelComposite`.

nim65s, 2019-07-17:
Thanks a lot !

closed 2019-08-07 by jmirabel