humanoid-path-planner/hpp-pinocchio #114: SEGV at data creation since v4.8.0 and pinocchio > v2.1.9

olivier-roussel, 2020-03-11:
As I am using both HPP and Pinocchio frameworks, I usually handle my device through an hpp::pinocchio::Device and if necessary manage separated datas (hpp::pinocchio::Data) for doing Pinocchio computations.
The following minimalist example was working fine with hpp-pinocchio v4.7.0 and pinocchio v2.1.9, but now SEGV at hpp::pinocchio::Data creation with update to hpp-pinocchio v4.8.0 and pinocchio v2.3.1.
```
#include <hpp/pinocchio/device.hh>
#include <hpp/pinocchio/joint-collection.hh>
#include <pinocchio/multibody/data.hpp>

int main(int /*argc*/, char** /*argv*/)
{
  auto robot = hpp::pinocchio::Device::create ("test");
  auto data = hpp::pinocchio::Data(robot->model()); // SEGV

  return 0;
}
```
If using only pinocchio interface everything is fine. But I still need to wrap with hpp-pinocchio at some point.

Is there something I am missing ?

jmirabel, 2020-03-11:
I tried your snippet and do not have any SEGV.

Mind that, on 18.04 and above, executable will find first the installed library, before the newly build library.

jmirabel, 2020-03-11:
Are you sure the executable is not using an old library ?

olivier-roussel, 2020-03-11:
Ok that's good news. I will double check that the proper library is linked, but I see no reason to that at first (excepted if lib names have changed). Also I'm running 16.04.

Digging a bit, that was the BT I got:
```
#0  __memcpy_avx_unaligned () at memcpy-avx-unaligned.S:114
#1  0x000000000048d113 in Eigen::internal::smart_copy_helper<double, true>::run (start=0x0, end=0x30, target=0x1fbadb0) at Memory.h:496
#2  0x00000000004892bd in Eigen::internal::smart_copy<double> (start=0x0, end=0x30, target=0x1fbadb0) at Memory.h:487
#3  0x00007ff2b6a5e689 in Eigen::DenseStorage<double, -1, -1, 1, 0>::DenseStorage (this=0x1fb9d78, other=...) at DenseStorage.h:506
#4  0x00007ff2b6a5727d in Eigen::PlainObjectBase<Eigen::Matrix<double, -1, 1, 0, -1, 1> >::PlainObjectBase (this=0x1fb9d78, other=...) at PlainObjectBase.h:514
#5  0x00007ff2b6a4d951 in Eigen::Matrix<double, -1, 1, 0, -1, 1>::Matrix (this=0x1fb9d78, other=...) at Matrix.h:370
#6  0x00007ff2b6b14141 in pinocchio::Tensor<double, 3, 0, long>::Tensor (this=0x1fb9d78, other=...) at tensor.hpp:151
#7  0x00007ff2b6b13339 in pinocchio::DataTpl<double, 0, hpp::pinocchio::JointCollectionTpl>::DataTpl (this=0x1fb9020) at data.hpp:29
#8  0x00007ff2b6b11d67 in hpp::pinocchio::DeviceData::DeviceData (this=0x1fb8f80, other=...) at device-data.cc:39
#9  0x00007ff2b6a41c1f in hpp::pinocchio::Device::numberDeviceData (this=0x1fb67e0, s=@0x7ffdc4495960: 1) at device.cc:84
#10 0x00007ff2b6a41687 in hpp::pinocchio::Device::Device (this=0x1fb67e0, name=...) at device.cc:58
#11 0x00007ff2b6a41d40 in hpp::pinocchio::Device::create (name=...) at device.cc:97
```

jmirabel, 2020-03-11:
So the SEGV is the the creation of the device, with AVX turned on. Could you tell me what option you turned on ? You may use `gcc <your_gcc_options> -E -v - </dev/null 2>&1 | grep cc1` to get the exhaustive list.

I think there is an issue in pinocchio or hpp-fcl related to this. I also got a SEGV when I activated AVX and tried to use pinocchio. I did not have time to dig further.

olivier-roussel, 2020-03-11:
Here is the output of ```gcc -E -v - </dev/null 2>&1 | grep cc1```:
```/usr/lib/gcc/x86_64-linux-gnu/5/cc1 -E -quiet -v -imultiarch x86_64-linux-gnu - -mtune=generic -march=x86-64 -fstack-protector-strong -Wformat -Wformat-security```

olivier-roussel, 2020-03-11:
Seems that as I was compiling with GCC 5, c++11 support is not enabled by default (this appeared in GCC 6 : https://gcc.gnu.org/gcc-6/changes.html ). And cmakelists does not enable or check c++11 support, so the whole framework was not compiled with c++11 support (which was not the desired behavior). Apparently, this lead to this bug (maybe because pinocchio was not using the Eigen Tensor module, but this is supposed to be supported) or maybe because Urdfdom must be compiled with c++11 as @jmirabel suggested.
Forcing c++11 support for HPP and Pinocchio frameworks fixed this. Maybe c++11 support should be forced (in cmake) or GCC version checked to avoid such cases, as default c++11 support is not standard for compilers.

closed 2020-03-11 by olivier-roussel

jmirabel, 2020-03-11:
Thanks for the feedback.