humanoid-path-planner/hpp-pinocchio #116: ManipulatorArm2 lost a hand ?

nim65s, 2020-04-14:
Hi,

Trying to run tests, I got this error in several places.
```
fatal error: in "difference_and_distance": std::invalid_argument: Model does not have any body named left_hand
```

https://github.com/humanoid-path-planner/hpp-pinocchio/commit/c4dcebfa05160e08d12072147c08082a35924db3 workaround this issue.

Did the manipulator really lost a hand ? Or is this something new where we can now easily change a hand, but sommething is missing here ?

jmirabel, 2020-04-14:
It loads file *package://hpp_environments/tests/baxter_simple.urdf*. Has the file changed ?
You may try to load it with pinocchio and see the list of joint names.

https://github.com/humanoid-path-planner/hpp-pinocchio/blob/f348551f3efb86e00df2a25ee725b3ee11482310/src/simple-device.cc#L126

nim65s, 2020-04-14:
This file never changed. It loads fine on pinocchio 2.3.1, and doesn't on 2.4.0.

I'll move the issue to pinocchio then, thanks !

closed 2020-04-14 by nim65s