humanoid-path-planner/hpp-pinocchio #119: Move Container<JointAndShapes_t> from hpp::manipulation::Device

florent-lamiraux, 2020-04-27:
I consider moving member
  `Container <JointAndShapes_t> jointAndShapes;`
from class `hpp::manipulation::Device` to `hpp::pinocchio::Device`. The goal is to ease the construction of `explicit_::ConvexShapeContact` class with a list of names of floor contact surfaces and a list of object contact surfaces.

This implies to move `Shape` and `JointAndShape` definitions to hpp-pinocchio.
Does anyone have an objection ?

jmirabel, 2020-04-27:
This also implies that the environment contact must be in the device. I think we don't want this. For the robot shapes, I agree with moving them to hpp-pinocchio.

If it is only for building the object, there is no need to modify the class itself. You can only add a function. I think it is better to keep the class independent when possible.

florent-lamiraux, 2020-04-27:
But since we have used pinocchio, have environment contact surfaces not been attached to the joint "universe" ?

florent-lamiraux, 2020-04-27:
I can maybe find a solution without changing classes, simply by moving simple type definitions to hpp-pinocchio.

jmirabel, 2020-04-27:
Since the environment is not in the robot, I don't see why the environment surfaces would be in the robot...

florent-lamiraux, 2020-04-27:
Looking into the code, it seems that ProblemSolver::jointAndShapes is not used: I couldn't find a line in core/problem-solver.cc nor in manipulation/problem-solver.cc that adds data to this member, whereas in hpp-manipulation-urdf/src/srdf/factories/contact.cc, all surfaces are added to the robot.

jmirabel, 2020-04-27:
Have a look at robot::loadEnvironmentModel in hpp-manipulation-corba.

florent-lamiraux, 2020-04-27:
I see. Environment contact surfaces are stored in object and copied in ProblemSolver.

closed 2020-05-05 by florent-lamiraux