humanoid-path-planner/hpp-pinocchio #132: C++ standard requirements

nim65s, 2020-09-28:
Hello,

In https://github.com/humanoid-path-planner/hpp-pinocchio/commit/659212f7e5e575c510d033bd284d639fe6ff07e2 I added a C++11 requirement in this package, because I saw some
`unknown location(0): fatal error in "serialization": std::logic_error: Not implemented without C++ 11.` here.

But after the 4.10 release, this error is still there in the tests: https://gitlab.laas.fr/humanoid-path-planner/hpp-pinocchio/-/jobs/93575

So I guess that actually, it's the hpp-util package that requires c++11, right ?

jmirabel, 2020-09-28:
No, this is a mistake at https://github.com/humanoid-path-planner/hpp-pinocchio/blob/8d9e202f0d004928bbdee290475d23ba61123376/src/device.cc#L538

I guess it should be `>=`.

closed 2020-10-15 by nim65s

jmirabel, 2020-10-15:
FYI, I probably made this mistake in other packages.

jcarpent, 2020-10-15:
I think these flags should be done once for all in the JRL module

commit 241719a0 (2020-09-29): Merge pull request #133 from nim65s/devel

Fix standard guard, fix #132
commit 507660bb (2020-09-29): fix standard guard, fix #132