humanoid-path-planner/hpp-pinocchio #146: pinocchio v2.6.0 incompatibility

nim65s, 2021-04-14:
Hi,

This is already fixed in devel, but in the current master and latest release:
```
[  2%] Building CXX object CMakeFiles/hpp-pinocchio.dir/src/collision-object.cc.o
…/src/collision-object.cc: In member function ‘hpp::fcl::CollisionObject* hpp::pinocchio::CollisionObject::fcl(hpp::pinocchio::GeomData&) const’:
…/src/collision-object.cc:72:25: error: ‘hpp::pinocchio::GeomData {aka struct pinocchio::GeometryData}’ has no member named ‘collisionObjects’; did you mean ‘collisionRequests’?
       return & geomData.collisionObjects[geomInModelIndex];
                         ^~~~~~~~~~~~~~~~
                         collisionRequests
…/src/collision-object.cc: In member function ‘const hpp::fcl::CollisionObject* hpp::pinocchio::CollisionObject::fcl(const GeomData&) const’:
…/src/collision-object.cc:77:25: error: ‘const GeomData {aka const struct pinocchio::GeometryData}’ has no member named ‘collisionObjects’; did you mean ‘collisionRequests’?
       return & geomData.collisionObjects[geomInModelIndex];
                         ^~~~~~~~~~~~~~~~
                         collisionRequests
…/src/collision-object.cc: In member function ‘hpp::fcl::CollisionObject* hpp::pinocchio::CollisionObject::fcl(hpp::pinocchio::DeviceData&) const’:
…/src/collision-object.cc:85:28: error: ‘hpp::pinocchio::GeomData {aka struct pinocchio::GeometryData}’ has no member named ‘collisionObjects’; did you mean ‘collisionRequests’?
       return & geomData(d).collisionObjects[geomInModelIndex];
                            ^~~~~~~~~~~~~~~~
                            collisionRequests
…/src/collision-object.cc: In member function ‘const hpp::fcl::CollisionObject* hpp::pinocchio::CollisionObject::fcl(const hpp::pinocchio::DeviceData&) const’:
…/src/collision-object.cc:91:28: error: ‘const GeomData {aka const struct pinocchio::GeometryData}’ has no member named ‘collisionObjects’; did you mean ‘collisionRequests’?
       return & geomData(d).collisionObjects[geomInModelIndex];
                            ^~~~~~~~~~~~~~~~
                            collisionRequests
…/src/collision-object.cc: In member function ‘void hpp::pinocchio::CollisionObject::move(const Transform3f&)’:
…/src/collision-object.cc:121:18: error: ‘hpp::pinocchio::GeomData {aka struct pinocchio::GeometryData}’ has no member named ‘collisionObjects’; did you mean ‘collisionRequests’?
       geomData().collisionObjects[geomInModelIndex]
                  ^~~~~~~~~~~~~~~~
                  collisionRequests
```

florent-lamiraux, 2021-04-14:
If you release the `devel` branch, what is the problem ?

nim65s, 2021-04-14:
@florent-lamiraux : I am working on the release. This issue is to explain the pinocchio v2.6.0 API break to @jcarpent, as he requested.

jcarpent, 2021-04-14:
Is the issue coming from Pinocchio?

closed 2021-05-10 by nim65s