humanoid-path-planner/hpp-pinocchio #205: Compilation error with Pinocchio 3

ymontmarin, 2024-07-09:
When trying to compile `hpp-pinocchio` with pinocchio 3.1.0, I encouter the following error:
```
CMake Error at /home/conda/feedstock_root/build_artifacts/hpp-pinocchio_1720511918692/_h_env_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placeho/lib/cmake/pinocchio/pinocchioTargets.cmake:79 (set_target_properties):
  The link interface of target "pinocchio::pinocchio_extra" contains:

    Qhull::qhullcpp

  but the target was not found.  Possible reasons include:

    * There is a typo in the target name.
    * A find_package call is missing for an IMPORTED target.
    * An ALIAS target is missing.

Call Stack (most recent call first):
  /home/conda/feedstock_root/build_artifacts/hpp-pinocchio_1720511918692/_h_env_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placeho/lib/cmake/pinocchio/pinocchioConfig.cmake:182 (include)
  cmake/package-config.cmake:113 (find_package)
  CMakeLists.txt:108 (add_project_dependency)
```
which may be related to the new behavior of pinocchio  that have multiple targets (pinocchio_default, pinocchio_parser, pinocchio_geometry, pinocchio_exta, pinocchio_paralelle) with pinocchio target being an alias for pinocchio_default, pinocchio_parser, pinocchio_geometry, pinocchio_exta et pinocchio_paralelle.

Do you know how I can fix this error ?

Also I note in CMakeLists.txt that we have `add_project_dependency(hpp-util 5.0.0 REQUIRED)`, shouldn't the version of hpp-util match the one of hpp-pinocchio ?

Thank you in advance for the help

jorisv, 2024-07-09:
A new version (build 3) of the [qhull package had been generated](https://github.com/conda-forge/qhull-feedstock/pull/17/files#diff-f3725a55bf339595bf865fec73bda8ac99f283b0810c205442021f29c06eea9a)

I think, by removing [static build](https://github.com/conda-forge/qhull-feedstock/pull/17/files#diff-d7075654874cb08007a21aaab3ecd4b3453a9087e7505d034d548b8938b599bcR8) and testing that [qhullcpp doesn't exists](https://github.com/conda-forge/qhull-feedstock/pull/17/files#diff-f3725a55bf339595bf865fec73bda8ac99f283b0810c205442021f29c06eea9aR35) this new package create the issue.

nim65s, 2024-07-09:
There is nothing in hpp-pinocchio about qhull.  So if you have this error, it comes from the packaging of pinocchio.

Can you double check the content of files under`lib/cmake/pinocchio`in your install prefix for anything related to qhull ? I can't reproduce right now, because I was not aware of this new `BUILD_WITH_EXTRA_SUPPORT` flag in pinocchio, which is required to enable qhull.

About the version number, this one can mean different things based on how the project is configured, ref. https://cmake.org/cmake/help/latest/module/CMakePackageConfigHelpers.html#generating-a-package-version-file. Here, we use the default from jrl-cmakemodules, which is `SameMajorVersion`:

https://github.com/jrl-umi3218/jrl-cmakemodules/blob/7faf6f9c4e981111bf0ac5dff4970f0959c11f69/package-config.cmake#L187-L195

So this `add_project_dependency(hpp-util 5.0.0 REQUIRED)` line means we require hpp-util with `5.0.0 <= version < 6`

jorisv, 2024-07-09:
I have create an issue on [qhull feedstock](https://github.com/conda-forge/qhull-feedstock/issues/18)

closed 2024-07-17 by ymontmarin